luc-github/Marlin #1: Feedback ESP3D for Marlin

luc-github, 2019-03-18:
New repository creation to sync with 2.X easily as now ESP32 code from @simon-jouet is part of Marlin as well as WebUI from @grownseed

So sync of ESP3D must be done on this base - as requested by @vivian-ng 
Unfortunately all previous comments are now lost because part of previous fork
This should allow to do a PR if any to Marlin more easily I hope.

Current temporary branch https://github.com/luc-github/Marlin/tree/ESP32_WEBUI based on previous fork will be merged to https://github.com/luc-github/Marlin/tree/bugfix-2.0.x this week  and ESP32_WEBUI branch will be deleted.

Once missing features will be ported from GRBL_ESP32 FW and ESP3D 3.0  FW and considered stable on Marlin :
* Telnet
* Bluetooth
* Pushover/Line/Email Notifications
* etc...

Then I will do a PR to Marlin FW with hope it can be accepted

luc-github, 2019-03-18:
Porting is done https://github.com/luc-github/Marlin/tree/bugfix-2.0.x
now testing and SD support looks broken ... same hardware is fine with  https://github.com/luc-github/Marlin/tree/ESP32_WEBUI 
will investigate what is broken

luc-github, 2019-03-19:
I wrote some instructions and listed new GCODe for wifi : https://github.com/luc-github/Marlin/wiki/Setup-for-ESP32
@vivian-ng let me know if it match your needs

vivian-ng, 2019-03-19:
@luc-github 
Thank you for merging the latest changes from Marlin/bugfix-2.0.x.

After compiling, there were a few problems.
1. My test board uses the old pin definitions for ESP32 (not the ones that @simon-jouet added, which uses GPIO expander pins that are not available on my bareboard test board). Not a big issue, I simply created a new pins file for my board. But when I tried to send a pull request to your repo, I couldn't find this repo from the list of base repos available. Not a pressing issue, though.
2. My motors hum and buzz the moment power is supplied to the board. And cannot be moved. When I rolled back to the previous version (pre-merge), the motors work fine, and do not hum/buzz. I am not sure what it is, but I suspect it could be how the HAL handles stepper motors (maybe there is a difference from the pre-merge version and the current Marlin/bugfix-2.0.x). Probably this merge [PoC ESP32 i2s dma](https://github.com/MarlinFirmware/Marlin/pull/12959).

The good thing is you fixed the SD support. Now, when I start, I don't need to send M21 manually before doing a refresh of the SD card.

luc-github, 2019-03-19:
1 - well you can PR to this repo, it should work, but you need to refork it as previous fork does not exists anymore so won't appear available.
2 - well for this part I cannot help a lot, I have the latest board from Simon (all in one) but I use the board only not a full system as my code do not touch any part of the "core" of ESP32 HAL / Marlin
<s>what do you suggest to do ?</s> just see your comment on Marlin, let see the feedback
3 - the auto M21 is new feature of 2.x not from me - but now need to explicitly set the SS pin in pin file unlike before, that was my issue.

Let me know if I can help on anything

vivian-ng, 2019-06-20:
@luc-github Will you be able to merge your current repo to Marlin bugsfix-2.0.x? There were some updates to the ESP32 HAL issues which I am hoping to test out.

Also, is there any way to get long filename support in the webUI? Marlin already has long filename support in it, hopefully the webUI can be updated to read long filenames too. Just a wish list.

I am also getting "Guru Meditation Error: Core 1 panic'ed (LoadProhibited)" error when I use M587 S"xxx" P"12345678" command on the terminal. This had never happened in the past, so I guess I will have to play around to see what changed to cause this error.

luc-github, 2019-06-20:
What update do you need ? I can sync easily in theory.
I need to see about filename. Marlin SD lib does not actually support long file name. Commands print use only short filenames or there was new SD lib integrated ?
Long filename wrote by PC can be retrieved but I am pretty sure I can write long file name with Marlin SD lib unless there was some update.
About crash I will check. What decoded stack says?

vivian-ng, 2019-06-20:
@luc-github Thanks for the quick reply.
Recently, there has been some work to get I2S stepper stream to work for the ESP32.
I need this [fix](https://github.com/MarlinFirmware/Marlin/pull/13743) and probably other related ones too. I have also added a new board definition, and will send you a pull request once ready.

vivian-ng, 2019-06-20:
`Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014fd  PS      : 0x00060c30  A0      : 0x800df79c  A1      : 0x3ffb1e20  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb1da0  
A10     : 0x3ffb1de4  A11     : 0x3ffb1e20  A12     : 0x00000000  A13     : 0x0000000b  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000006  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400014fd:0x3ffb1e20 0x400df799:0x3ffb1e30 0x400d3b00:0x3ffb1e50 0x400e3ad0:0x3ffb1f30 0x400e3b9e:0x3ffb1f50 0x400e5005:0x3ffb1f70 0x400e0995:0x3ffb1f90 0x400fc451:0x3ffb1fb0 0x40088649:0x3ffb1fd0
`
This is the error I get when using M587.

luc-github, 2019-06-20:
You need to use stack decoder to give readable output of this stack

luc-github, 2019-06-20:
Ok looks there are some conflict that prevent the merging from upstream - I will work on it this week end

vivian-ng, 2019-06-21:
@luc-github Thank you for looking into this.

As for the issue with M587, I am getting problems with M589 too. I think this could be a gcode parsing error?
Anyway, here is the decoded stack:
```
0x400df799: WiFiConfig::isSSIDValid(char const*) at /tmp/arduino_build_987379/sketch/src/HAL/HAL_ESP32/wificonfig.cpp line 112
0x400d3b00: GcodeSuite::M587() at /tmp/arduino_build_987379/sketch/src/HAL/HAL_ESP32/gcode_ESP32.cpp line 210
0x400e3ad0: GcodeSuite::process_parsed_command() at /tmp/arduino_build_987379/sketch/src/gcode/gcode.cpp line 653
0x400e3b9e: GcodeSuite::host_keepalive() at /tmp/arduino_build_987379/sketch/src/gcode/gcode.cpp line 850
0x400e5005: advance_command_queue() at /tmp/arduino_build_987379/sketch/src/gcode/queue.cpp line 880
0x400e0995: _GLOBAL__sub_I_card() at /home/user/.arduino15/packages/esp32/hardware/esp32/1.0.3-rc1/cores/esp32/IPAddress.h line 94
0x400fc451: dtostrf at /home/user/.arduino15/packages/esp32/hardware/esp32/1.0.3-rc1/cores/esp32/stdlib_noniso.c line 120
0x40088649: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

luc-github, 2019-06-21:
May be , to reproduce issue 
M587 S"xxx" P"12345678", you really put "xxx"?  or another SSID ?

vivian-ng, 2019-06-21:
`M587 S"xxx" P"12345678"`
I put in my SSID and password for xxx and 12345678. I tried both with and without quotation marks but I get errors both ways.

luc-github, 2019-06-21:
and same SSID was used and ok before ?

vivian-ng, 2019-06-21:
@luc-github Please don't worry about this issue. It could be something on my end, I don't think it is a problem with the code, since it worked before, and the web UI works to allow me to enter my wifi credentials.

luc-github, 2019-06-21:
I will check after merging done anyway - just wanted all info to reproduce ^_^

vivian-ng, 2019-06-22:
@luc-github 
For long filename support in the web interface, one possibility is to use [M33](http://marlinfw.org/docs/gcode/M033.html) gcode to convert names obtained via [M20](http://marlinfw.org/docs/gcode/M020.html). Internally, though, the filenames will still have to be stored within the interface in DOS 8.3 format since [M23](http://marlinfw.org/docs/gcode/M023.html) and other SD commands use that format. I don't think anyone wants to rewrite the files in the `Marlin/src/sd` subdirectory to allow Marlin to natively work in long-filename mode instead of DOS 8.3 format. 😅

BTW, is the SD card listing function part of the ESP3D firmware (C/C++ code), or is it part of the ESP3D-WEBUI portion (HTML/JavaScript)?

luc-github, 2019-06-22:
@vivian-ng I am well aware of current Marlin limitations that is why I said:  

*  Marlin commands only support shorname (print / delete/ etc...) that is why M20 use only short names.   
* Long file names wrote by PC can be retrieved, it can be used by M33. To answer your last question, currently I do not rely on M20, my ESP3D porting generate the files list using same API as M20, I have already prepared possible support for long name, I just do not currently query the long name using Marlin API for each short name provided, but as I wrote I can be done.   
*  The upload is the actual real limitation: write has only short name support, and I do not see any way to do it but using ESP32 SD library which is not compatible with Marlin SD lib, and I have no plan to spend time on this as 8.3 is a Marlin `feature`   : https://github.com/MarlinFirmware/Marlin/issues/6165

To sum up: 
1 - Display long names can be done doing extra actions for files copied from PC
2 -Upload/write does not support long names
Conclusion:
Display long name is useless on uploaded files - so what the point to do the extra actions if it is useless ?        
Additionnaly IMHO It will be confusing for end user to see long names for files manually copied and short names for files uploaded. 

What do you think?

luc-github, 2019-06-22:
I did a rebase and fix compilations issue due to new code : https://github.com/luc-github/Marlin/tree/premerge-bugfix2-0.x
I did not tested yet but compilation and flash is Ok - will test webUI tomorow and rebase bugfix2-0.x in same way if everything is ok

vivian-ng, 2019-06-23:
Thank you @luc-github for helping with the rebase!
Is there anyway for me to help with rebase? My repo is a fork of your repo, which is a fork of the Marlin repo... how do you try to merge with the upstream of an upstream, and iron out issues with conflicts?

luc-github, 2019-06-23:
@vivian-ng  I rebased my fork and fixed conflicts manually by keeping my changes instead of conflicted ones, these conflicts are due to changes that are related to the current webUI already merged.
Then I fixed compilation issues / some code due to changes which are still related to merged WebUI and some change in Marlin API for injecting GCODE command.

In your case you ,must rebase and do same I guess, using Github Desktop App and Atom it is pretty easy, it was my first time that is why I wanted to try on separate branch  first, but was easy so if I succeed you should also ^_^

I will redo same on my bugfix2-0.x today after some tests on premerge branch so you can rebase your bugfix2-0.x from my bugfix2-0.x, ping me on gitter if you need some help

.

luc-github, 2019-06-23:
@vivian-ng I found the root cause of M587 issue - stupid memory allocation release, I have also fixed a bug not allowing to save empty password for open network.
The M503 has new behavior in output so also need to update webUI - need at least 2.1b41, I have put it in data directory
I just finished tests and did rebase on bugfix2-0.x branch -  seems ok on my side
Have fun ^_^

vivian-ng, 2019-06-24:
@luc-github 
The following #include statement in `Marlin/src/HAL/HAL_ESP32/HAL.cpp` causes problems because there is no `spiffs.h` in the folder.
```
#if EITHER(EEPROM_SETTINGS, WEBSUPPORT)
  #include "spiffs.h"
#endif
```
I could compile when I commented it out.

I2S also doesn't seem to be working. I am going to give stock Marlin a try later to see if it works.

luc-github, 2019-06-24:
@vivian-ng  Strange I may miss it in 2nd rebase because these define are not enabled - removed from code - thank you

About I2S there are now 2 init , and I have added a define as you did before may be the define is not enabled on your side.
```
void HAL_init(void) {
#if ENABLED(I2S_STEPPER_STREAM)
    i2s_init();
#endif
}
void HAL_init_board(void) {
  #if ENABLED(WIFISUPPORT)
    wifi_config.begin();
  #endif
}
```

Also I just pushed a FR on Marlinto see if there could have and alternative : https://github.com/MarlinFirmware/Marlin/issues/14390

 this would simplify the futur sync and allow to get Marlin in normal way instead of rebasing fork

vivian-ng, 2019-06-24:
@luc-github You are my hero!
Thank you for reminding me to check my `#define I2S_STEPPER_STREAM` statements.

I had used
```
#ifdef I2S_STEPPER_STREAM
  #undef I2S_STEPPER_STREAM
#endif
```
in my non-I2S board definition.
I changed that to:
```
#ifdef I2S_STEPPER_STREAM
  #define I2S_STEPPER_STREAM
#endif
```
for the board definition with I2S. Which obviously didn't work.
Changed
`#ifdef I2S_STEPPER_STREAM`
to
`#ifndef I2S_STEPPER_STREAM`
and it is working now!

As for your FR on Marlin repo, I am not very sure what you are trying to say. I am not very good with git and all that fork, sync, rebase, merge, etc. terminology, so I will read your FR again to try and figure out what it means.

luc-github, 2019-06-24:
Good you figured out
About Feature Request to simplify. Use Marlin and set a define and download ESP3D for Marlin lib and just put it in libraries directory and you get all working

luc-github, 2019-06-24:
So any update from Marlin are take in account if you sync Marlin repo. For ESP3D sync just sync with ESP3D lib repo

vivian-ng, 2019-07-17:
Hi @luc-github, I think the ESP32 HAL guys have made some updates to the HAL on the main Marlin repo, will you be syncing your fork with the main repo in the near future?

Also, I really hope you can consider pushing your changes (especially everything in the HAL_ESP32 directory that does not require changes to other parts of Marlin) back to the main repo. Marlin is a community effort and the recent response I see from the ESP32 HAL guys show that they welcome your efforts. You are making Marlin better, so I don't see why anyone will have objections. The Marlin lead may be too busy now preparing for 2.0 RC1, maybe that is why he has yet to reply on a "policy" or "direction" but I really don't think that should stop you, since you are not breaking anything.

And even for changes that are in other parts of Marlin outside the HAL_ESP32 directory, as long as they are encapsulated inside `#ifdef ESP32` statements, I don't think it will be blocked by the rest of the Marlin people.

For consideration, please.

luc-github, 2019-07-17:
@vivian-ng I have done the rebase and fixed the conflicts and compilation issue
* moved your pins_MRR_ESP.h to esp32 pins dir
* changed the board number to 6001 following new ESP32 board numbering
* added I2S #define in your pin file but I let you adjust the values, I have put the esp32 ones for the moment, without them compilation failed

```
//
// I2S (steppers & other output-only pins)
//
#define I2S_STEPPER_STREAM
#define I2S_WS              25
#define I2S_BCK             26
#define I2S_DATA            27
```

About change yes I am considering pushing them that is why I have opened a issue in Marlin repo, but my fork remove the existing WebUI code and its files are not compatible with existing files, so it is mandatory to get a decision to make either cohexistence ( add tons of #ifdef in Hal, and move files to different subdirectories),or do a complet replacement ,or allow  external lib.
I can definitly work on one solution once I get answer ^_^, but not before.

Also looking at the recent updates on ESP32 per git history, it seems people are available to read PR and merging them so I will wait to get answer ^_^

vivian-ng, 2019-07-17:
`get a decision to make either cohexistence ( add tons of #ifdef in Hal, and move files to different subdirectories),or do a complet replacement ,or allow external lib.`

Which option would make things easiest for you? Which option would you prefer?
1. Coexistence: Sounds like the easiest option to get buy-in from the Marlin people, but seems like the most work on your part.
2. Replacement: Probably the easiest for you, but we will need to convince the Marlin people that it will not break other parts of Marlin (support for other, non-ESP32 boards).
3. External lib: To me, this sounds like a good compromise, except in the long-term, trying to maintain what is essential two versions of the ESP32 HAL will be quite a bit of work.

Personally, I think the changes you have made (EEPROM support, webserver/webUI, OTA) are good enough to replace the existing Marlin features for these. But let's see what the rest have to say.

luc-github, 2019-07-17:
I am fine with any solution - of course replace all would be easier but allowing external lib would give more flexibility IMHO.
Well let see what is the decision, if any decision popup ;)

Let me know if the rebase is ok on your side

luc-github, 2019-07-19:
@vivian-ng FYI I got one issue with latest rebase,  if no temp sensor is connected then boot crash on temperature in dead loop, but no issue with 999 sensor as I do not have full system but just @simon-jouet 's board.

This issue come with latest Dynamic ADC attenuation I guess, as I did not had this issue before.

Luc

vivian-ng, 2019-07-21:
@luc-github Thank you for the rebase. I have tried compiling it, and I don't have the crashing problem without thermistor. But in the past, I always get a "thermal runaway" halt when I flash firmware onto a bare ESP32 module.

I am also getting very bad quality prints when I print a stringing test. I don't know if it has to do with temperature or filament. I will try to print something later to see if a different filament works better.

luc-github, 2019-07-21:
@vivian-ng happy to read you do not have issue, so it may be my board not ready for this

Hope you will figure out what is the print issue

luc-github, 2019-08-04:
@vivian-ng are you still on basic MRR ESPA ?   
 I read here http://teck78.blogspot.com/2019/07/mrr-espa-basic-esp32-based-3d-printer.html you plan to move to i2S stepper feature , did you succeed ? 
I see you board still use 4 steppers https://github.com/vivian-ng/Hardware/issues/2

vivian-ng, 2019-08-04:
@luc-github 
I am still working on the MRR ESPA, which is no I2S, but that is almost done.
So I am concurrently working on the I2S version, MRR ESPE. Right now, I have a board layout for the I2S version with 4 shift registers to test out I2S features. So far, the stepper motors are working on I2S, and PWM seems to work (I have yet to actually test it out on a heated bed or hotend; I just connected to a LED to see if I can control brightness).
The I2S version also needs a bit of work on layout... very messy traces now all over the place. I need to rearrange the placing of the components a bit. Design consideration is to fit within 102mm by 102mm so that I can make use of cheap PCB manufacturing for prototype. Anything larger than that, JLCPCB will charge me extra instead of USD2.

But all these will have to wait a week... still overseas. :)

luc-github, 2019-08-04:
@vivian-ng Ok than you, so ESPE is the advanced you mentioned, understood.  
enjoy your trip ^_^

vivian-ng, 2019-09-03:
@luc-github Hi, I don't know if you have any experience with this, but there are times when I send commands during a print via the webUI, and the printer would hang with this:
`Error 0 :`

That's all, no other messages.

This happened with babystepping (I use a macro, which send M290 Z0.01), which I at first thought was a problem with Marlin. But it also happened when I tried to change the flow rate via the webUI (which sends a M221 gcode, I think), so I was wondering if it could be something else, like how gcode sent over the terminal is causing some kind of error with Marlin. Do you have any idea which file to look at to see how gcode sent from the webUI is being handled by the underlying Marlin firmware?

luc-github, 2019-09-03:
@vivian-ng `Error 0:` means no answer or connection error

does `M290 Z0.01` send any answer from serial ?

vivian-ng, 2019-09-03:
@luc-github When I am printing something, and a `M290 Z0.01` gets sent, the reply is (after "busy processing") `Error 0 :`. Same with trying to change the flow rate via the webUI (which sends a `M221 Sxx`), I also get `Error 0 :`.

When I am not printing anything, there is no response to `M290 Z0.01` (I don't think the firmware responds to babystepping when not printing). Using the webUI to change flow rate, etc. I get the normal `Ok` response. Plus when I check (for example, `M221`), I can see the flow rate has been changed.

luc-github, 2019-09-03:
I will check if I can workaround the no answer from Marlin and avoid the Error 0 with web command by faking it

luc-github, 2019-09-03:
@vivian-ng I cannot reproduce the issue changing the flow rate - I got Ok answer from FW
[image]

I have enabled also BABYSTEPING 
and sent the command  `M290 Z0.01` and same - no error 0
[image]

Can you use the latest webUI - the one I currently use to see if is an old issue already solved : https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz

luc-github, 2019-09-04:
@vivian-ng any update ?
I can only reproduce issue when board is power off which is normal

vivian-ng, 2019-09-04:
@luc-github Thank you for testing. Can I check if you are sending the gcodes/using the webUI while the board is printing something? Because when the board is not actively running a print, the commands get through. I only get the `Error 0 :` issue when the board is printing something, and then I try to do babystepping or use the webUI to change flow.

BTW, the UI version I have says it is 2.1b41.

luc-github, 2019-09-04:
@vivian-ng sorry  I missed this part, I missread your comment  
I do not have a full system, so did not try when pinting , just a  board-
In that case I think I know what the problem is,  I think it is because webserver is only handled when  idle, I was expecting idle happen when all actions in loop are done  before doing a new loop. 
Ok so it is not in web-UI then

Solution is to not use idle task but create a dedicated task for webserver and it should be ok,

I cannot work on it today but I should be able to have something tomorrow,

luc-github, 2019-09-04:
@vivian-ng can you give a try to https://github.com/luc-github/Marlin/commit/f9d5b73c22e7a9e1b9d84fed9b7849849e1e25ba
I moved the web handle task from idle  task to specific task and pinned it to core 0 so it won't disturb core 1 and normal Marlin functions

vivian-ng, 2019-09-05:
@luc-github Thank you for making the changes.

I tested the webUI using the new commits. This test was conducted while printer is idle.
This is the terminal output when I used the arrows to move the X and Z axis. Seems to be okay.
```
G91
G1 X10 F1000
G90
Ok
eecho:enqueueing "G1 X10 F1000"
echo:enqueueing "G90"
echo:enqueueing "M114"
X:7.00 Y:-20.00 Z:0.00 E:0.00 Count X:559 Y:-1601 Z:0
G91
G1 Z1 F100
G90
Ok
eecho:enqueueing "G1 Z1 F100"
echo:enqueueing "G90"
echo:enqueueing "M114"
X:7.00 Y:-20.00 Z:1.00 E:0.00 Count X:559 Y:-1601 Z:401
```

This is the output when I clicked on auto-check temperature, seems to be broken.
```
oekcho T::enqueueing "M105"
54.47 /0.00 B:47.22 /0.00 @:0 B@:0
Ok
echoo:kenq uTeue:ing "M10554"
.33 /0.00 B:47.11 /0.00 @:0 B@:0
Ok
eochko: Ten:queueing "54.0M105"5
 /0.00 B:47.19 /0.00 @:0 B@:0
Ok
eokcho T::enqueueing "53.91M105"
 /0.00 B:46.96 /0.00 @:0 B@:0
Ok
echoo:kenq uTeue:ing "M10553"
.82 /0.00 B:46.99 /0.00 @:0 B@:0
```
The temperature graph also does not show, as I think the webUI is unable to get the necessary info from the serial terminal.

Auto-check position also gives weird readings.
```
X:echo:7en.q0ue0ueing  Y":-20.00M114"
 Z:3.00 E:0.00 Count X:559 Y:-1601 Z:1203
Ok
X:echo:e7nque.u0ei0ng " Y:-M11420"
.00 Z:3.00 E:0.00 Count X:559 Y:-1601 Z:1203
Ok
echo:enqueueing ":M114"
7.00 Y:-20.00 Z:3.00 E:0.00 Count X:559 Y:-1601 Z:1203
```
When I try to use the webUI to start a print:
```
echo:enqueueing "echMo3 /1PETG.GCO":
Now fresh file: /1PETG.GCO
echo:enqueueing "M24"
open failed, File: 1PETG.GCO.
echeo:cNhoow :freshenq ufeiuleei:n g /1PETGF.GCO
"M23 /1PETGF.GCO"
echo:enqueueing "M24"
open failed, File: 1PETGF.GCO.
```
I am not able to use M23 to select the file too.
```
M23 /1PETG.GCO
Ok
eechoch:oNo:w enfqrueesuhei nfgi le": /1PETG.GCO
M23 /1PETG.GCO"
open failed, File: 1PETG.GCO.
M23 1PETG.GCO
Ok
echeoc:ho:enNqouwe uefirnegs h" file: M23 1PETG.GCO"1PETG.GCO
open failed, File: 1PETG.GCO.
```

I can't test if the `Error 0 :` still occurs since I cannot start a print.

luc-github, 2019-09-05:
@vivian-ng it seems command are corrupted using separated task - I am back home and now can reproduce issue so I work on it

luc-github, 2019-09-05:
@vivian-ng I think it is fixed - I changed the web command using `GCodeQueue::enqueue_one_now` by filling the web socketbuffer which I was expecting to be slower but it solve the issue per my tests

I have also fixed web-UI displaying 2 temperatures output instead of one, and the Printer settings was broken due to missing "\n" in answer, the latest webUI index.html.gz is in data directory

I have simulate a print by using a gcode file moving extruder and I can get answer now when Marlin send the answer
Let me know if it is ok on your side

vivian-ng, 2019-09-05:
@luc-github Thank you for making the changes! I have tested it on a dry run too and it seems to work. I don't get `Error 0 :` anymore when I try to use `M290 Z0.01`. I will try to find time to connect the board up to an actual printer tomorrow to see if it can print.

The temperature graph does not update during `M190` or `M109` though. Probably because Marlin blocks commands while waiting for temperature in these two commands... will it be possible to use `M155` for auto report of temperatures? That way, Marlin will continue to report temperature when while waiting for temperature to be reached when using `M190` or `M109`. Just a suggestion, not a game breaker.

luc-github, 2019-09-05:
Added M155 auto report support in latest webUI : https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz

vivian-ng, 2019-09-06:
@luc-github Thank you for adding the M155 support. It works! And I like the simple but elegant way it was done.

I have tested the updated firmware with an actual print. It is able to accept commands via webUI now even when printing.

Even babystepping via M290 seems to work during prints (when not using I2S stepper stream).

luc-github, 2019-09-06:
great ^_^

vivian-ng, 2019-10-15:
I am getting reboots when using the firmware with I2S stepper stream on a coreXY setup. I think the issue has to do with I2S implementation (and have raised this issue [here](https://github.com/simon-jouet/ESP32Controller/issues/10#issuecomment-540326134)), but I can't be sure as WiFi related function calls are also being made in the decoded stack trace.

```
PC: 0x400f418e: Stepper::endstop_triggered(AxisEnum) at Marlin/src/module/stepper.cpp line 2228
EXCVADDR: 0x00000000

Decoding stack results
0x400f418e: Stepper::endstop_triggered(AxisEnum) at Marlin/src/module/stepper.cpp line 2228
0x400f2aba: Planner::endstop_triggered(AxisEnum) at Marlin/src/module/planner.cpp line 1489
0x400f15a6: Endstops::update() at Marlin/src/module/endstops.cpp line 706
0x400f1613: Endstops::poll() at Marlin/src/module/endstops.cpp line 268
0x400f509f: Temperature::isr() at Marlin/src/module/temperature.cpp line 2800
0x400f50bf: tempTC_Handler() at Marlin/src/module/temperature.cpp line 2324
0x40080fa5: timer_isr(void*) at Marlin/src/HAL/HAL_ESP32/HAL_timers_ESP32.cpp line 75
0x40089d2d: vTaskExitCritical at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 4274
0x4008babf: multi_heap_internal_unlock at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap.c line 380
0x4008c144: multi_heap_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c line 202
0x40084a74: heap_caps_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c line 111
0x40084aa9: heap_caps_malloc_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c line 140
0x4008618d: _malloc_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/syscalls.c line 37
0x4016a7af: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 50
0x4016a781: operator new[](unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc line 32
0x400fac07: WiFiUDP::parsePacket() at /home/user/data/platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiUdp.cpp line 210
0x400ff61e: ArduinoOTAClass::handle() at /home/user/data/platformio/packages/framework-arduinoespressif32/libraries/ArduinoOTA/src/ArduinoOTA.cpp line 382
0x400e1056: WiFiServices::handle() at Marlin/src/HAL/HAL_ESP32/wifiservices.cpp line 133
0x400e0cdd: WiFiConfig::handle() at Marlin/src/HAL/HAL_ESP32/wificonfig.cpp line 423
0x400d311b: HAL_idletask() at Marlin/src/HAL/HAL_ESP32/HAL.cpp line 84
0x400e1358: idle(bool) at Marlin/src/Marlin.cpp line 697
0x400f2bae: Planner::synchronize() at Marlin/src/module/planner.cpp line 1541
0x400f1e77: do_homing_move(AxisEnum, float, float) at Marlin/src/module/motion.cpp line 1249
0x400f1f3a: homeaxis(AxisEnum) at Marlin/src/module/motion.cpp line 1435
0x400e3295: GcodeSuite::G28(bool) at Marlin/src/gcode/calibrate/G28.cpp line 327
0x400e4b0b: GcodeSuite::process_parsed_command(bool) at Marlin/src/gcode/gcode.cpp line 244
0x400e52ad: GcodeSuite::process_next_command() at Marlin/src/gcode/gcode.cpp line 830
0x400e695c: GCodeQueue::advance() at Marlin/src/gcode/queue.cpp line 880
0x400e15c5: loop() at Marlin/src/Marlin.cpp line 1159
0x40107529: loopTask(void*) at /home/user/data/platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp line 19
0x400884a5: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```
Do you think I am right to assume this is a I2S implementation issue?
Maybe possible problems with HAL_timers_ESP32.cpp? Seems like the HAL_timer_enable_interrupt and HAL_timer_disable_interrupt don't do anything.
```
/**
 * Enable timer interrupt on the timer
 * @param timer_num timer number to enable interrupts on
 */
void HAL_timer_enable_interrupt(const uint8_t timer_num) {
  //const tTimerConfig timer = TimerConfig[timer_num];
  //timer_enable_intr(timer.group, timer.idx);
}

/**
 * Disable timer interrupt on the timer
 * @param timer_num timer number to disable interrupts on
 */
void HAL_timer_disable_interrupt(const uint8_t timer_num) {
  //const tTimerConfig timer = TimerConfig[timer_num];
  //timer_disable_intr(timer.group, timer.idx);
}
```
Or possible problem with stepper.cpp? Because it says something like this:
```
void Stepper::stepper_pulse_phase_isr() {
...
    // TODO: need to deal with MINIMUM_STEPPER_PULSE over i2s
    #if MINIMUM_STEPPER_PULSE && DISABLED(I2S_STEPPER_STREAM)
      // Just wait for the requested pulse duration
      while (HAL_timer_get_count(PULSE_TIMER_NUM) < pulse_end) { /* nada */ }
    #endif
```
I don't understand Marlin's way of planning the steppers enough, so it is tough for me to debug this on my own.

luc-github, 2019-10-15:
yes previous decoded stack show ISR issue.

if you have some doubt about wifi disable it and send G28 by serial.

luc-github, 2019-10-15:
about Marlin code and I2S I did not look at it yet.
I am just back home today so I may be able to have a look only later this week

vivian-ng, 2019-10-16:
Without WiFi (M588 S0 P1)
```
Guru Meditation Error: Core  1 panic'ed (Coprocessor exception)
Core 1 register dump:
PC      : 0x400f418e  PS      : 0x00060931  A0      : 0x800f2abd  A1      : 0x3ffbe620  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x3f000000  A5      : 0x00000000  
A6      : 0x637f0000  A7      : 0xcc800000  A8      : 0x800f4188  A9      : 0x3ffc3080  
A10     : 0x00000000  A11     : 0x3ffb0060  A12     : 0x40083907  A13     : 0x3ffbe7c0  
A14     : 0x02ffffff  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x00000004  
EXCVADDR: 0x00000000  LBEG    : 0x40002390  LEND    : 0x4000239f  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x400f418e  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40082a51

Backtrace: 0x400f418e:0x3ffbe620 0x400f2aba:0x3ffbe700 0x400f15a6:0x3ffbe720 0x400f1613:0x3ffbe740 0x400f509f:0x3ffbe760 0x400f50bf:0x3ffbe780 0x40080fa5:0x3ffbe7a0 0x40081859:0x3ffbe7c0 0x40083904:0x3ffb1d80 0x4008362b:0x3ffb1da0 0x4008132f:0x3ffb1dc0 0x400ee6f3:0x3ffb1de0 0x400ee9a0:0x3ffb1e00 0x400e1343:0x3ffb1e30 0x400f2bae:0x3ffb1e50 0x400f1e77:0x3ffb1e70 0x400f1f3a:0x3ffb1ed0 0x400e3295:0x3ffb1f00 0x400e4b0b:0x3ffb1f30 0x400e52ad:0x3ffb1f50 0x400e695c:0x3ffb1f70 0x400e15c5:0x3ffb1f90 0x40107529:0x3ffb1fb0 0x400884a5:0x3ffb1fd0

PC: 0x400f418e: Stepper::endstop_triggered(AxisEnum) at Marlin/src/module/stepper.cpp line 2228
EXCVADDR: 0x00000000

Decoding stack results
0x400f418e: Stepper::endstop_triggered(AxisEnum) at Marlin/src/module/stepper.cpp line 2228
0x400f2aba: Planner::endstop_triggered(AxisEnum) at Marlin/src/module/planner.cpp line 1489
0x400f15a6: Endstops::update() at Marlin/src/module/endstops.cpp line 706
0x400f1613: Endstops::poll() at Marlin/src/module/endstops.cpp line 268
0x400f509f: Temperature::isr() at Marlin/src/module/temperature.cpp line 2800
0x400f50bf: tempTC_Handler() at Marlin/src/module/temperature.cpp line 2324
0x40080fa5: timer_isr(void*) at Marlin/src/HAL/HAL_ESP32/HAL_timers_ESP32.cpp line 75
0x40083904: esp_timer_impl_get_time at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/esp_timer_esp32.c line 159
0x4008362b: esp_timer_get_time at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/esp_timer.c line 515
0x4008132f: millis at /home/user/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-misc.c line 139
0x400ee6f3: MarlinUI::update_buttons() at Marlin/src/lcd/ultralcd.cpp line 1133
0x400ee9a0: MarlinUI::update() at Marlin/src/lcd/ultralcd.cpp line 762
0x400e1343: idle(bool) at Marlin/src/Marlin.cpp line 666
0x400f2bae: Planner::synchronize() at Marlin/src/module/planner.cpp line 1541
0x400f1e77: do_homing_move(AxisEnum, float, float) at Marlin/src/module/motion.cpp line 1249
0x400f1f3a: homeaxis(AxisEnum) at Marlin/src/module/motion.cpp line 1435
0x400e3295: GcodeSuite::G28(bool) at Marlin/src/gcode/calibrate/G28.cpp line 327
0x400e4b0b: GcodeSuite::process_parsed_command(bool) at Marlin/src/gcode/gcode.cpp line 244
0x400e52ad: GcodeSuite::process_next_command() at Marlin/src/gcode/gcode.cpp line 830
0x400e695c: GCodeQueue::advance() at Marlin/src/gcode/queue.cpp line 880
0x400e15c5: loop() at Marlin/src/Marlin.cpp line 1159
0x40107529: loopTask(void*) at /home/user/platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp line 19
0x400884a5: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

I guess it has to do with the I2S stepper stream. Somehow, the endstop is interfering with the planner (which it should not).

luc-github, 2019-10-16:
yes at least it narrow down where need to investigate

luc-github, 2019-10-19:
@vivian-ng I will try to reproduce later today. I do not have an I2S system because I was focusing on ESP3D part
but I will have a look to ISR issue

luc-github, 2019-10-19:
@vivian-ng if line 2228 `if (was_enabled) DISABLE_STEPPER_DRIVER_INTERRUPT();` do nothing I doubt it is the root cause
I suspect more :
```
endstops_trigsteps[axis] = 0.5f * (
      axis == CORE_AXIS_2 ? CORESIGN(count_position[CORE_AXIS_1] - count_position[CORE_AXIS_2])
                          : count_position[CORE_AXIS_1] + count_position[CORE_AXIS_2]
    );
```
using float calculation on isr to fill an int32_t is not best I think

may be try :
```
endstops_trigsteps[axis] =  (
      axis == CORE_AXIS_2 ? CORESIGN(count_position[CORE_AXIS_1] - count_position[CORE_AXIS_2])
                          : count_position[CORE_AXIS_1] + count_position[CORE_AXIS_2]
    ) / 2;
```
also try to add `IRAM_ATTR` to be sure it can be call fast from ISR call
`void IRAM_ATTR Stepper::endstop_triggered(const AxisEnum axis)`

Just some ideas

[35 more replies]

closed 2019-11-28 by luc-github