luc-github/Marlin #2: Add missing features

luc-github, 2019-03-19:
* Merge Telnet Support from GRBL_ESP32, ESP3D 3.0
* Merge Bluetooth Support from GRBL_ESP32, ESP3D 3.0
* Merge Notification Support from GRBL_ESP32, ESP3D 3.0
* TBD

luc-github, 2019-06-28:
I will wait feedback from Marlin developers before merging the missing features, because it doe not make sense to add them if no chance to get some integration or if no one need ESP3D in Marlin
https://github.com/MarlinFirmware/Marlin/issues/14390

vivian-ng, 2019-07-13:
```
I will wait feedback from Marlin developers before merging the missing features, because it doe not make sense to add them if no chance to get some integration or if no one need ESP3D in Marlin
```
I just want to say I appreciate the work you have been doing. Thank you.
I have added to the conversation at the main repo, and linked that FR to another discussion related to the ESP32 HAL, so let's see if others working on the HAL are willing to join in.

luc-github, 2019-07-13:
Thank you for your support ^_^, on my side I won't push more as I did, the issue is open on Marlin waiting for feedback.
In same time, no worry I will continue to sync  Marlin 2.0 FW with this fork on demand according your needs