luc-github/Marlin #9: ESP32 V-Plotter

phantomse, 2019-07-03:
In my search for an hanging V-Plotter I wanted to make it wireless (Battery or Power-line only)

ESP3D continues figuring out the wireless part and can be added as library in Marlin.
There is a project which has all steps needed to change in Marlin (old version) for an V-Plotter  https://github.com/RickMcConney/PenPlotter 
Only since Marlin 2.0 there is support for esp32 so there may be some changes.

Hardware:
UNO D1 R32 (only one pin seams to be incompatible, snippet it off) + CNC Shield V3.0
LCD + Stepper + Servo + Solenoid + Endstops + SDCard ... => auto color changing and advanced functions
all in one, similar to https://hackster.imgix.net/uploads/attachments/536161/img_0131_joZbk8YliZ.JPG?auto=compress%2Cformat&w=1280&h=960&fit=max

Managed to compile and flash D1 R32 with Atom and PlatformIO but connecting to AP Marlin_ESP doesn't give me a Popup window
Is MarlinESP3D at a usable stage for my Project ( send .gcode to SD ovrer wifi, ... ) Cant find much documentation on the progress.

luc-github, 2019-07-03:
Hi good question - I did not listed the ported feature but only the one missing
here you have : https://github.com/luc-github/Marlin/issues/10
I am trying to sync with Marlin 2.0 time to time but some feature are little bit different than Marlin 2.0
like EEPROM saving Marlin use SPIFFS on Marlin 2.0 when I use EEPROM feature. 

About popup window not apearing - I guess you are talking about the captive portal
this feature is little bit peaky depending operating system - this is ESP32 limitation not ESP3D, if when your join ESP3D in AP mode your OS does not launch the browser - type the ip of your ESP3D which is by default : 192.168.0.1

Let me know if the sum up I did is clear enough for you

phantomse, 2019-07-03:
Only Marlin 2.1 will get polar kinematic ( still years ... )
https://github.com/MarlinFirmware/Marlin/issues/5275
there are also old versions of Repetier and GRBL with polar kinematic but non esp32 compatible.
https://github.com/jasonwebb/grbl-mega-wall-plotter
https://github.com/RickMcConney/PenPlotter
Polargraph is trying to do all from scratch on ESP32 and thereby has little progress.
We have the hardware only the software lags behind.

luc-github, 2019-07-03:
Are you sure for GRBL_32 ? https://github.com/bdring/Grbl_Esp32/tree/Wall_Plotter
I know @bdring has some polar project and use GRBL_ESP32 : https://github.com/bdring/Polar-Coaster
they use also ESP3D ^_^

luc-github, 2019-07-03:
as you mentioned Marlin 2.0 does not fit your needs and so opened issue in grbl_esp32 I guess 
|issue can be closed

closed 2019-07-03 by luc-github

bdring, 2019-07-03:
I have a wall plotter project I am using to learn about the Trinamic drivers. There is a branch, but it is incomplete. It is a back burner project that only gets attention as time allows. The Trinamic SPI features are in the master branch though. Stepper motors are not ideal for battery powered projects, but the Trinamic drivers have features that help. Coolstep allows you to set a max current, but the motors will use less when the load is light. They also can have a lower standby current when idle.

My plotter is pretty unique. It stores the belt on spools that are wound via constant force springs harvested from measuring tapes. Using the Trinamic StallGaurd feature it can home itself and determine the distance between motors. This allows the machine to be setup at any width and still be accurate.

It uses (2) controller boards. One is fully populated and one just gets a few connectors and is mounted inverted.

[image]

i-make-robots, 2019-08-19:
I love the constant force springs.  My attempts to use bearings as you have done (keep the belt on the pulley) all resulted in lost steps.  I get how you can use the stallguard as homing switch, but how does it tell you the distance between motors?

bdring, 2019-08-19:
One of the sides has a socket for the end of the other belt. You home that belt with it installed in the socket. Then you home it again with it loose. It knows how far it had to go to home it the second time. There is a calibration value that stores the length that needs to be added to get the actual center to center dimension. Stallgaurd is accurate to about 1 full step of the motor.

The stallgaurd makes it possible to home anywhere. That is key to the system. The back off after homing makes it easy to release the belt. The wall plotter branch of Grbl_ESP32 has the homing code. Any kinematics in there are untested.

I am working on a more universal kinematics system for Grbl_ESP32. The wall plotter project is on hold until that gets done.

[image]

i-make-robots, 2019-08-19:
I love it!  I couldn't see that from the other side.  Do you spring load the bearing that tensions the belt on the pulley?  We failed to get it to tension consistently and it sometimes lost steps.

bdring, 2019-08-19:
The socket was not in the earlier pictures.

I have not had any lost steps yet when fully assembled. I have seen a few skips if there is loose belt on the take up side, but the spring loaded reels appear to fix that. Are you using 2 bearings?

There is no spring loading on the bearings. There are slotted holes and I just push it hard against the belt before tightening.

[image]

i-make-robots, 2019-08-19:
I used a single rollerskate bearing on each side.

i-make-robots, 2019-08-19:
Ugh, it so good looking.  I want to incorporate every good idea into the Makelangelo.