luc-github/Marlin #13: compiling errors

bob808, 2019-11-03:
Hi,
I'm having problems compiling. Arduino IDE doesn't work, either on 1.8.10 or 1.9 beta. Can't remember the error but the guys working on Marlin recommended using platformio.
I also tried the official marlin 2.0 with platformio and it compiles ok.
But this fork I'm having troubles with even in platformio.

### Steps to Reproduce
install platformio/add Atmel SAM platform and compile

**Expected behavior:** [What you expect to happen]
finish compiling
**Actual behavior:** [What actually happens]
errors out
#### Additional Information
`.pioenvs/DUE_USB/src/src/libs/clib/u8g_virtual_screen.c.o: warning: multiple common of `u8g_vs_height'
.pioenvs/DUE_USB/src/src/HAL/HAL_DUE/U8glib-HAL-master/src/clib/u8g_virtual_screen.c.o: warning: previous
common is here
.pioenvs/DUE_USB/src/src/libs/clib/u8g_virtual_screen.c.o: In function `u8g_SetVirtualScreenDimension':
C:\Users\Bob\Desktop\Marlin-bugfix-2.0.x/Marlin\src\libs\clib/u8g_virtual_screen.c:126: multiple definition of `u8g_SetVirtualScreenDimension'
.pioenvs/DUE_USB/src/src/HAL/HAL_DUE/U8glib-HAL-master/src/clib/u8g_virtual_screen.c.o:C:\Users\Bob\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_DUE\U8glib-HAL-master\src\clib/u8g_virtual_screen.c:126: first defined here
.pioenvs/DUE_USB/src/src/libs/clib/u8g_virtual_screen.c.o: In function `u8g_AddToVirtualScreen':
C:\Users\Bob\Desktop\Marlin-bugfix-2.0.x/Marlin\src\libs\clib/u8g_virtual_screen.c:133: multiple definition of `u8g_AddToVirtualScreen'
.pioenvs/DUE_USB/src/src/HAL/HAL_DUE/U8glib-HAL-master/src/clib/u8g_virtual_screen.c.o:C:\Users\Bob\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_DUE\U8glib-HAL-master\src\clib/u8g_virtual_screen.c:133: first defined here
.pioenvs/DUE_USB/src/src/libs/clib/u8g_virtual_screen.c.o:(.data.u8g_dev_vs+0x0): multiple definition of `u8g_dev_vs'
.pioenvs/DUE_USB/src/src/HAL/HAL_DUE/U8glib-HAL-master/src/clib/u8g_virtual_screen.c.o:(.data.u8g_dev_vs+0x0): first defined here
.pioenvs/DUE_USB/src/src/libs/clib/u8g_virtual_screen.c.o:C:\Users\Bob\Desktop\Marlin-bugfix-2.0.x/Marlin\src\libs\clib/u8g_virtual_screen.c:48: multiple definition of `u8g_vs_cnt'
.pioenvs/DUE_USB/src/src/HAL/HAL_DUE/U8glib-HAL-master/src/clib/u8g_virtual_screen.c.o:C:\Users\Bob\Desktop\Marlin-bugfix-2.0.x/Marlin\src\HAL\HAL_DUE\U8glib-HAL-master\src\clib/u8g_virtual_screen.c:48: first defined here
.pioenvs/DUE_USB/src/src/gcode/gcode.cpp.o: In function `GcodeSuite::process_parsed_command(bool)':
C:\Users\Bob\Desktop\Marlin-bugfix-2.0.x/Marlin\src\gcode/gcode.cpp:682: undefined reference to `GcodeSuite::M586()'
C:\Users\Bob\Desktop\Marlin-bugfix-2.0.x/Marlin\src\gcode/gcode.cpp:681: undefined reference to `GcodeSuite::M585()'
C:\Users\Bob\Desktop\Marlin-bugfix-2.0.x/Marlin\src\gcode/gcode.cpp:683: undefined reference to `GcodeSuite::M587()'
C:\Users\Bob\Desktop\Marlin-bugfix-2.0.x/Marlin\src\gcode/gcode.cpp:684: undefined reference to `GcodeSuite::M588()'
C:\Users\Bob\Desktop\Marlin-bugfix-2.0.x/Marlin\src\gcode/gcode.cpp:685: undefined reference to `GcodeSuite::M589()'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\DUE_USB\firmware.elf] Error 1
===================================== [FAILED] Took 2917.55 seconds =====================================

Environment           Status    Duration
--------------------  --------  ------------
megaatmega2560        IGNORED
megaatmega1280        IGNORED
at90usb1286_cdc       IGNORED
at90usb1286_dfu       IGNORED
DUE                   IGNORED
DUE_USB               FAILED    00:48:37.549
DUE_debug             IGNORED
LPC1768               IGNORED
LPC1769               IGNORED
melzi                 IGNORED
melzi_optiboot        IGNORED
rambo                 IGNORED
sanguino_atmega644p   IGNORED
sanguino_atmega1284p  IGNORED
STM32F1               IGNORED
fysetc_STM32F1        IGNORED
BIGTREE_SKR_MINI      IGNORED
STM32F4               IGNORED
STM32F7               IGNORED
ARMED                 IGNORED
alfawise_U20          IGNORED
mks_robin             IGNORED
mks_robin_mini        IGNORED
mks_robin_nano        IGNORED
JGAURORA_A5S_A1       IGNORED
black_stm32f407ve     IGNORED
BIGTREE_SKR_PRO       IGNORED
teensy35              IGNORED
malyanm200            IGNORED
esp32                 IGNORED
fysetc_f6_13          IGNORED
linux_native          IGNORED
================================= 1 failed, 0 succeeded in 00:48:37.549 =================================`

I have two kinds of errors in the "Problems" tab. 
All are related to Configuration.h file.
Sample of errors:
`{
	"resource": "/c:/Users/Bob/Desktop/Marlin-bugfix-2.0.x/Marlin/Configuration.h",
	"owner": "C/C++",
	"severity": 8,
	"message": "function call is not allowed in a constant expression",
	"startLineNumber": 167,
	"startColumn": 5,
	"endLineNumber": 167,
	"endColumn": 12
}`
`{
	"resource": "/c:/Users/Bob/Desktop/Marlin-bugfix-2.0.x/Marlin/Configuration.h",
	"owner": "C/C++",
	"severity": 8,
	"message": "expected an expression",
	"startLineNumber": 1070,
	"startColumn": 33,
	"endLineNumber": 1070,
	"endColumn": 34
}`
[image]

luc-github, 2019-11-03:
this fork is for ESP32 board why do you use due ?

bob808, 2019-11-03:
I thought it is for ESP32 enabled marlin builds. like using esp3d with marlin

luc-github, 2019-11-03:
There is not such thing sorry -  this fork is for Marlin on ESP32 board itself = 1 board like https://github.com/maplerainresearch/MRR_ESPA or https://github.com/simon-jouet/ESP32Controller/tree/r2

I do not test Due on this fork so Due code is not as up to date - I just rebase it when ESP32 code change on Marlin 2.0 so I am behind lastest update.

to use an ESP32 board and ESP3D with  with a due board (= 2 boards), you need 2 FW : Marlin for Due from Malrin github and ESP3D 2.1 from ESP3D github.

bob808, 2019-11-03:
Ah silly me. I was reading around this repo late at night and my neurons might have fallen asleep.  
So then, thank you for your work! Sure is nice.

closed 2019-11-03 by bob808