luc-github/Marlin #14: Feedback ESP3D for Marlin part2

luc-github, 2019-11-28:
follow #1 as now too long

luc-github, 2019-11-28:
Quick update on rebase which is harde than expected.
 rebase is done but got  some issues that are under investigation

* WebUI need update : done

* if enabling lcd compilation failed under arduino IDE but now ok under platformIO

* using Arduino IDE EPPROM acccess is Ok (M500 succeed) but failed under PlatformIO
I have also experienced failure to connect to my router in sta mode which may related to my tests corrupting more the NVS 
I did not implemented a reset function on Marlin unlike other ESP3D porting  that may now become necessary to implement ASAP to solve such issues

on ESPE:
* CR10 screen is working (when connecting with proper connector 👿 )
* rotary encoder work partially on my side : push = ok +/- = not working, I did not investigated yet
* beep is not working neither  I did not investigated yet

on ESPA:
no test yet with latest rebase

will continue debug tomorrow

vivian-ng, 2019-11-28:
@luc-github Before the rebase, I was able to use LCD controller including beep with the MRR ESPE, so I will test this out again over the weekend.

BTW, for beep, remember to put the BEEPER jumper.

vivian-ng, 2019-11-29:
@luc-github I am trying to do a test compile, but I am running into errors.

I did a `git fetch upstream` and a `git merge upstream/bugfix-2.0.x`, then manually updated the conflicts to accept the incoming changes.

I tried to compile, but after several attempts, have been unsuccessful.

In HAL_ESP32
`watchdog.h` seems to be missing a `#endif` at the end of the file.
There are also double declarations of the `tTimerConfig` structure in `timers.h` and `HAL_timers_ESP32.h`, seems that some files use the HAL version of the header, some CPP files just use `timer.h`.

I guess I will wait for you to give the go ahead before I try to compile again.

luc-github, 2019-11-29:
@vivian-ng  after a rebase git fetch upstream and a git merge upstream/bugfix-2.0.x, will go wrong  on sub fork
check my repo e.g. watchdog.h is correct, and no more HAL_XXX_ESP32 files 
https://github.com/luc-github/Marlin/tree/bugfix-2.0.x/Marlin/src/HAL/HAL_ESP32

you may refork my repo to have clean environment rebase break subfork sometimes

About beeper pin, yes jumper is there

luc-github, 2019-11-29:
Ok I did a simple test : use native code of Marlin 2.0- as for encoder ESP3D is not related 
change  board to BOARD_ESPRESSIF_ESP32 update RX/TX values
Change BOARD_ESPRESSIF_ESP32 pins by ESPE ones and compile

lcd works but not rotary as in rebase so issue is in Marlin and no due to rebase - will check what changed later

vivian-ng, 2019-11-29:
I did a test compile for both MRR ESPA and MRR ESPE, no issues with compiling.

Enabling CR10STOCKDISPLAY in Configuration.h also allowed me to use my Creality Ender-3 LCD controller with no issue. I have yet to test things through, though. Will do so tomorrow when I do a proper compile to make sure it can run my Ender-3 using a MRR ESPE.

Over the next few days or so, I intend to:
1. Test on Ender-3 on MRR ESPE running TMC2130 SPI on X and Y axes.
2. Test on coreXY printer using MRR ESPA running TMC2130 SPI on X and Y axes.

And I will try to find time for to test out various combinations of I2S/native pins for stepper control to see if they can be mixed.

`platformio.ini` needs to be tweaked to allow TMC library to be properly found and compiled. I have put up a PR for that, which also tries to bring the `platformio.ini` closer to what is found on stock Marlin 2.0.

luc-github, 2019-11-29:
@vivian-ng you mean on your side rotaty and beeper are working  with my latest  https://github.com/luc-github/Marlin/tree/bugfix-2.0.x   ?

vivian-ng, 2019-11-29:
@luc-github Beeper is working now... due to a stupid mistake on my part, got the pin def for the beeper pin mixed up. It is in the PR too.

luc-github, 2019-11-29:
Yes I confirm beeper is now ok 
but rotary encoder turn left / turn right is not, but revert to old Marlin code and it works, so pins are ok and hardware on my side
do you still confim turn left / turn right works on your side ?

vivian-ng, 2019-11-29:
I can use the LCD controller to make menu selections and such, so I think it is working. But my board is not connected to any motors now. It is something that I will test when I have time later next week. Priority is to do a test print on my Ender-3 tomorrow running on MRR ESPE, and try out MRR ESPA on my coreXY (right now, not connected).

luc-github, 2019-11-29:
please zip your FW sources that work for MRR_ESPE and push it here

I need to check something

luc-github, 2019-11-29:
Ok I found the root cause - it is a commit that happenened  1rst of  September, I found the solution and  also found another platform has the same issue and solved in same way also in September... so much code to read an test ...
I will push the fix later - I am releived I did not do / see something wrong 😸

vivian-ng, 2019-11-30:
I am having problems with manual mesh bed leveling. When I go to the menu, select "Level Bed", the screen does not change on the LCD. On serial terminal, I get this:
```
[13:15:55:520] echo:busy: processing
[13:15:57:520] echo:busy: processing
[13:15:59:520] echo:busy: processing
[13:16:01:520] echo:busy: processing
[13:16:01:984] X:0.00 Y:-25.00 Z:0.00 E:0.00 Count A:-1980 B:1980 Z:0
[13:16:01:987] ok
[13:16:01:987] Mesh Bed Leveling OFF
[13:16:01:998] 7x7 mesh. Z offset: 0.00000
[13:16:01:998] Measured points:
[13:16:01:998]         0        1        2        3        4        5        6
[13:16:01:998]  0 -0.17000 -0.06000 +0.11000 +0.15000 +0.04000 -0.08000 -0.22000
[13:16:02:004]  1 -0.10000 +0.04000 +0.14000 +0.18000 +0.12000 -0.02000 -0.15000
[13:16:02:011]  2 -0.05000 +0.08000 +0.22000 +0.26000 +0.16000 +0.06000 -0.12000
[13:16:02:017]  3 -0.05000 +0.12000 +0.17000 +0.20000 +0.16000 +0.06000 -0.10000
[13:16:02:023]  4 -0.08000 +0.03000 +0.12000 +0.16000 +0.12000 +0.02000 -0.16000
[13:16:02:028]  5 -0.20000 -0.08000 +0.02000 +0.06000 +0.02000 -0.10000 -0.26000
[13:16:02:034]  6 -0.26000 -0.11000 +0.02000 +0.06000 +0.02000 -0.12000 -0.28000
[13:16:02:039] 
[13:16:02:039] X:0.00 Y:-25.00 Z:0.00 E:0.00 Count A:-1980 B:1980 Z:0
[13:16:02:044] ok
```
The "busy processing" is when the printer is trying to home. But it seems that once the printer homes, it thinks it has completed the leveling process, and outputs the mesh (the mesh shown is the one stored in eeprom).

This is LCD related. I am able to do manual mesh bed leveling using gcode.

luc-github, 2019-11-30:
@vivian-ng 
Menu -> Motion ->Bed Leveling -> Level Bed ?
It is working on my side ^_^

Also I would like to suggest to  remove 
`#define REVERSE_MENU_DIRECTION` from pin files it is present on Configuration.h and with my screen it is not necessary so it could be confusing if I did not know it was there

luc-github, 2019-11-30:
Ok I revert the fix as it bring an issue - no more lcd when power off / power on ....

luc-github, 2019-12-01:
Ok I reverted the revert ^_^, I unconnected all from board and flashed again the boot is now ok
I connected all back but USB and screen is now also ok when start - seems lcd is acting strangely when booting with USB connected.
Screen is ok on rebase and rotary is ok with fix. Seems rotary issue is linked to lcd brand which is not an original CR10 but a clone

vivian-ng, 2019-12-02:
@luc-github Now that Marlin 2.0.0 is officially released, will you be basing future updates on 2.0.x or the bugfix branch?

My apologies but I have yet to test out the LCD code, will try and find time over the next few days (hopefully tonight) to do so.

luc-github, 2019-12-03:
@vivian-ng will be 2.0.x or dev-2.1.x or bugfix-2.0.x depending of your needs for next rebase

vivian-ng, 2019-12-03:
@luc-github I tested out the LCD code, it works with both original Ender-3 LCD controller as well as the clone. I also managed to get manual mesh bed leveling working. It was a mistake on my part, I did not enable `LCD_BED_LEVELING` in `Configuration.h`.

> will be 2.0.x or dev-2.1.x or bugfix-2.0.x depending of your needs for next rebase

I guess the rebase should be based on whichever is the main development branch for 2.0.x (the one that people are pushing changes to).

Also, I have updated `pins_MRR_ESPE.h` and `platformio.ini`, please see the PR. BTW, the `env:esp32` portion of stock Marlin's `platformio.ini` should work, and I think we should keep to that as much as possible for future integration with stock Marlin repo. My two cents worth opinion for your consideration.

luc-github, 2019-12-03:
Ho great - that is good news - I was afraid this change will break original CR10 

the change I did were to make my config working as I flashed it with Arduino IDE which set some things in board according these flags (board_build.flash_mode = qio)

ESP32 has lot of flags in ESP3D I set them all to avoid any `confusion` due to unset https://github.com/luc-github/ESP3D/blob/3.0/platformio.ini#L18-L32

I saw also issue on GRBL_ESP32 by not setting some flags https://github.com/bdring/Grbl_Esp32/issues/213#issuecomment-530769319

I have merged your PR, put things in comments is Ok in case of issue

Just one thing `monitor_speed = 115200`
by default Marlin is 250000 which is not consistent IMHO but I do not use Monitor so I am fine ^_^

vivian-ng, 2019-12-07:
@luc-github Thank you for merging the PR.

I was testing babystepping using LCD today on the MRR ESPE (I2S). Got an error with the following:
```
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x40089e15  PS      : 0x00060b34  A0      : 0x80174374  A1      : 0x3ffbe690  
A2      : 0x00000000  A3      : 0x3ffb8058  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060b21  A7      : 0x00000000  A8      : 0x3ffb8058  A9      : 0x0000abab  
A10     : 0x00000003  A11     : 0x00060b23  A12     : 0x00060b21  A13     : 0x00060921  
A14     : 0x00060721  A15     : 0x3ffbec24  SAR     : 0x00000019  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x400d3371  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40089e15

Backtrace: 0x40089e15:0x3ffbe690 0x40174371:0x3ffbe6b0 0x400d6e9b:0x3ffbe6e0 0x400f6fe7:0x3ffbe710 0x400e1bba:0x3ffbe730 0x400e1bdd:0x3ffbe750 0x400f7f25:0x3ffbe770 0x400f7f4b:0x3ffbe790 0x40081011:0x3ffbe7b0 0x400848a5:0x3ffbe7d0 0x400f2423:0x3ffb1cc0 0x400f0908:0x3ffb1cf0 0x400f091c:0x3ffb1d20 0x400f16ee:0x3ffb1d40 0x400e152b:0x3ffb1d70 0x400f6195:0x3ffb1d90 0x400f62fd:0x3ffb1dc0 0x400f6347:0x3ffb1e00 0x400f48a3:0x3ffb1e40 0x400f49ad:0x3ffb1eb0 0x400f4a78:0x3ffb1ef0 0x400e82f7:0x3ffb1f10 0x400e7193:0x3ffb1f30 0x400e79b1:0x3ffb1f50 0x400e9120:0x3ffb1f70 0x400e17a1:0x3ffb1f90 0x4010a469:0x3ffb1fb0 0x40088cb1:0x3ffb1fd0

Core 1 stack decode
-------------------
PC: 0x40089e15: vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 4274
EXCVADDR: 0x00000000

Decoding stack results
0x40089e15: vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 4274
0x40174371: timer_get_counter_value at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/timer.c line 54
0x400d6e9b: HAL_timer_get_count(unsigned char) at Marlin/src/HAL/HAL_ESP32/timers.cpp line 147
0x400f6fe7: Stepper::babystep(AxisEnum, bool) at Marlin/src/module/stepper.cpp line 2406
0x400e1bba: Babystep::step_axis(AxisEnum) at Marlin/src/feature/babystep.cpp line 47
0x400e1bdd: Babystep::task() at Marlin/src/feature/babystep.cpp line 56
0x400f7f25: Temperature::tick() at Marlin/src/module/temperature.cpp line 2731
0x400f7f4b: tempTC_Handler() at Marlin/src/module/temperature.cpp line 2339
0x40081011: timer_isr(void*) at Marlin/src/HAL/HAL_ESP32/timers.cpp line 75
0x400f2423: ftostr54sign(float const&, char) at Marlin/src/libs/numtostr.cpp line 244
0x400f0908: _lcd_babystep(AxisEnum, char const*) at Marlin/src/lcd/menu/menu_tune.cpp line 68
0x400f091c: _lcd_babystep_z() at Marlin/src/lcd/menu/menu_tune.cpp line 102
0x400f16ee: MarlinUI::update() at Marlin/src/lcd/ultralcd.h line 487
0x400e152b: idle() at Marlin/src/Marlin.cpp line 652
0x400f6195: Planner::_buffer_steps(XYZEval  const&, float, unsigned char, float const&) at Marlin/src/module/planner.h line 567
0x400f62fd: Planner::buffer_segment(float const&, float const&, float const&, float const&, float const&, unsigned char, float const&) at Marlin/src/module/planner.cpp line 2572
0x400f6347: Planner::buffer_line(float const&, float const&, float const&, float const&, float const&, unsigned char, float) at Marlin/src/module/planner.h line 661
0x400f48a3: segmented_line_to_destination(float const&, float) at Marlin/src/module/planner.h line 692
0x400f49ad: prepare_move_to_destination_cartesian() at Marlin/src/module/motion.cpp line 853
0x400f4a78: prepare_move_to_destination() at Marlin/src/module/motion.cpp line 1033
0x400e82f7: GcodeSuite::G0_G1() at Marlin/src/gcode/motion/G0_G1.cpp line 105
0x400e7193: GcodeSuite::process_parsed_command(bool) at Marlin/src/gcode/gcode.cpp line 229
0x400e79b1: GcodeSuite::process_next_command() at Marlin/src/gcode/gcode.cpp line 882
0x400e9120: GCodeQueue::advance() at Marlin/src/gcode/queue.cpp line 645
0x400e17a1: loop() at Marlin/src/Marlin.cpp line 1137
0x4010a469: loopTask(void*) at /home/user/platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp line 19
0x40088cb1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

Using babystepping by gcode gives the same error, minus the LCD portion of the stack.

This is likely an I2S issue since I tried gcode babystepping on the MRR ESPA (non-I2S) and it works.

luc-github, 2019-12-07:
@vivian-ng Ok will have a look seems timer problem again

luc-github, 2019-12-07:
@vivian-ng my ender clone just died, I got several time unresponsive and after restart  was ok. but not anymore...
can you give the gcode sequence you use for babystepping?

vivian-ng, 2019-12-07:
@luc-github There gcode I use:
`M290 Z-0.01`
and
`M290 Z0.01`
for babystepping down and up respectively.

I actually use the ESP3D and set up macros for these two. Then, after starting a print, I use the macros on the webUI to send these codes.

luc-github, 2019-12-09:
I have sent 10 times the babystep command `M290 Z0.01`  when printing and no crash
[image]
then same with  `M290 Z-0.01` 
[image]
and still no crash

My printer is a cartesian printer FYI

Do you have more details how to reproduce issue ?

vivian-ng, 2019-12-09:
My babystepping test on MRR ESPE:
- Ender-3 (cartesian)
- TMC2130 SPI mode on X and Y axes; A4988 on Z and E0
- CR10STOCKDISPLAY

The above gave me issues with babystepping. Both when using LCD controller and when using M290 gcode. Attached are my config files.
[configs.zip](https://github.com/luc-github/Marlin/files/3939085/configs.zip)

On MRR ESPA (non-I2S), running on coreXY printer with TMC2130 SPI mode on X and Y axes; A4988 on Z and E0, no errors when I use M290 gcode.

luc-github, 2019-12-09:
On EPSE 
My TMC2208 are in standalone and used everywhere
No Screen 
I will check your config but I think I already merge your config with the one you sent by PM

vivian-ng, 2019-12-12:
Just something to share.
**Do not set temperature on ESP3D webUI, then try to upload a file to the SD card.** It seems Marlin will be so "obsessed" with the upload, and forget to check if the hotend temperature is reached. It was lucky I was nearby (waiting to load filament) and saw the smoke and managed to turn off the printer in time.

luc-github, 2019-12-12:
Hi ok I will open issue for this - I will try if to have separate task for web solve the issue

vivian-ng, 2019-12-12:
As [previously mentioned](https://github.com/luc-github/Marlin/issues/1#issuecomment-550082971), there are random pauses in the prints, sometimes a whole print taking 8 hours can complete without a pause. At other times, a 2-hour print can pause twice.

[image]

You can see the two blobs here, where the print paused. The thing is, there is no message on serial terminal. The only symptom is that when the printer "pauses", I cannot send any commands to it via serial terminal (USB) or webUI. But the commands being sent via USB and webUI are buffered, it seems, because once the printer resumes printing, it will execute the commands (in my case, I just send M503 to test for response, and the printer will send back my settings once printing resumes).

luc-github, 2019-12-12:
need to add some log to SD for this I think
you confirm this happen when printing from SD ?

vivian-ng, 2019-12-13:
@luc-github Yes, all printing via SD card.

luc-github, 2019-12-13:
@vivian-ng  I did a branch with all wifi stuff moved to a separate task and and pinned to core 0
https://github.com/luc-github/Marlin/tree/test-task
I cannot deeply test as my espe is no more heating 😭 - I sent you PM about it

luc-github, 2019-12-14:
@vivian-ng please give try to https://github.com/luc-github/Marlin/tree/test-task - WDT is fixed, I have successfully uploaded/ downloaded when heating and heat is still under control
One limitation when uploading / dowloading if `M155` is not enabled the ui is not updated as pool queries are on hold until upload/download is done  , but Marlin is still in control of temperature so  when upload is back you can see it